jesserizzo/envoy_reader #1: Only supports the Envoy S, not original Envoy

DavidDeSloovere, 2018-08-22:
Hello @jesserizzo 
I was happy when I saw this new component in Home Assistant, until I figured out that my Envoy (original, not the S) does not have the json endpoint.
It only has HTML, but does include the 4 production values at "/production?locale=en"

[image]

Do you think it would be possible to support this 'scraping' or 'parsing' of the HTML?
I don't know enough Python to tackle this (or any for that matter).
But I can help with testing.

```
<table>
 <tr><td colspan="3">System has been live since
 <div class=good>Tue Feb 16, 2016 03:42 PM CET</div></td></tr>
 <tr><td>Currently</td>    <td>   634 W</td></tr>
<tr><td>Today</td>     <td> 6.39 kWh</td></tr>
<tr><td>Past Week</td>    <td> 73.3 kWh</td></tr>
<tr><td>Since Installation</td>    <td> 8.74 MWh</td></tr>
 </table>
```

Don't know what the best approach is for choosing between json or html. 
Maybe check if the .json endpoint returns a 301 to the error page (which is what I see) it might be a way to decide to switch to HTML. 
Maybe add an option like 'gateway support json' (or doesn't support json).
If there is something in Python to just get the value via a selector like `tr[1]td[1]`, it might not be that much work (notice if and might)

DavidDeSloovere, 2018-08-24:
Consumption is definitely not available. This envoy can only measure production.

This is awesome... http://envoy/api/v1/production does spit out **JSON**!

```
{
  "wattHoursToday": 879,
  "wattHoursSevenDays": 70522,
  "wattHoursLifetime": 8749128,
  "wattsNow": 1066
}
```

And http://envoy/api/v1/consumption returns a 404 with this body
{
  "status": 404,
  "error": "",
  "info": "Resource /api/v1/consumption not found",
  "moreInfo": ""
}

Regarding the update interval: both /api/v1 json data and /production html table update at the same time for me.

akwan, 2018-08-24:
@DavidDeSloovere does this show anything?

http://envoy/backbone/application.js

https://thecomputerperson.wordpress.com/2016/08/03/enphase-envoy-s-data-scraping/ has a lot of interesting references - wondering what else may be available.

Might be possible to impart some logic that can detect the version of the envoy and only allow features supported per said envoy.

jesserizzo, 2018-08-24:
@akwan 

That site is specifically about the Envoy S, so it probably won't help in this case.

Also, I had no idea you could just go to http://envoy
How does that even work? Is this reliable enough that I could not require the users to put in their IP address?

jesserizzo, 2018-08-24:
@DavidDeSloovere 

Try the newest commit.

akwan, 2018-08-25:
@jesserizzo Wasn't sure what the original envoy had and hadn't seen documentation for it so wasn't sure what it would've had.

I wouldn't recommend relying on a hostname sussed from multicast DNS. IP or user designated hostname is still the way to go. I keep my Envoy S IQ in a network separate from where I access it from, so I personally don't use http://envoy or http://envoy.local. I was just using https://envoy as a placeholder per @DavidDeSloovere's URLs.

https://enphase.com/en-us/support/how-do-i-view-current-production-my-envoy-s

DavidDeSloovere, 2018-08-25:
I was using http://envoy also as a placeholder, funny that it actually works.

Fix the redirect check in the PR #2 - you might want to check if it still works for your envoy S
auto redirect was on by default, so you get a 200 after a second internal request to the error page
it's first actual python I wrote, would have tried to optimize like I wrote in the PR, but maybe you want to do that yourself? just let me know

jesserizzo, 2018-08-25:
@DavidDeSloovere

Looks good, thanks.

Does waiting on the 301 error take too long? Otherwise I would probably leave it as is.

DavidDeSloovere, 2018-08-25:
Not that it's taking too long (not that I noticed), just not necessary. Might even optimize further for the *consumption* stat, by not even making the call in case of the original envoy. If I feel like writing some more python, I'll see what I can do later in another PR later.

jesserizzo, 2018-08-25:
@DavidDeSloovere 
Ok, can you try the newest commit. Thanks for your help.

@akwan 
Thanks for the info. Does your Envoy IQ support the [/production.json](url) endpoint?

akwan, 2018-08-25:
Yep, including /home.json and /inventory.json.

Before this integration I was using this from @fredsmith

https://git.smith.bz/derf/homeautomation/blob/master/sensors/solar.yaml

closed 2018-08-26 by jesserizzo

jesserizzo, 2018-08-27:
@DavidDeSloovere 
I already have a pull request in for home assistant. I'm not sure what you mean by that second part, so definitely make whatever changes you're thinking of.