jesserizzo/envoy_reader #5: Add support for gathering individual inverter production

gtdiehl, 2019-04-06:
I have an original Envoy and have been gathering inverter data from:
`http://<ip>/api/v1/production/inverters`

Unfortunately this page requires authentication, I use HTTPDigestAuth to gather the inverter data
`requests.get(ENPHASE_URL_INVERTERS, auth=HTTPDigestAuth(ENPHASE_USERNAME, ENPHASE_PASSWORD))`

The envoy returns data formatted as such (I changed my serial #'s):
`[{'serialNumber': '000000000000', 'lastReportDate': 1554513164, 'lastReportWatts': 22, 'maxReportWatts': 234}, {'serialNumber': '000000000001', 'lastReportDate': 1554513167, 'lastReportWatts': 22, 'maxReportWatts': 237}, {'serialNumber': '000000000002', 'lastReportDate': 1554513172, 'lastReportWatts': 19, 'maxReportWatts': 221}, {'serialNumber': '000000000003', 'lastReportDate': 1554513161, 'lastReportWatts': 22, 'maxReportWatts': 240}, {'serialNumber': '000000000004', 'lastReportDate': 1554513169, 'lastReportWatts': 19, 'maxReportWatts': 220}, {'serialNumber': '000000000005', 'lastReportDate': 1554513172, 'lastReportWatts': 19, 'maxReportWatts': 222}, {'serialNumber': '000000000006', 'lastReportDate': 1554513162, 'lastReportWatts': 24, 'maxReportWatts': 249}, {'serialNumber': '000000000007', 'lastReportDate': 1554513169, 'lastReportWatts': 19, 'maxReportWatts': 219}, {'serialNumber': '000000000008', 'lastReportDate': 1554513159, 'lastReportWatts': 24, 'maxReportWatts': 250}, {'serialNumber': '000000000009', 'lastReportDate': 1554513164, 'lastReportWatts': 24, 'maxReportWatts': 248}, {'serialNumber': '000000000010', 'lastReportDate': 1554513166, 'lastReportWatts': 19, 'maxReportWatts': 220}, {'serialNumber': '000000000011', 'lastReportDate': 1554513171, 'lastReportWatts': 19, 'maxReportWatts': 220}, {'serialNumber': '000000000012', 'lastReportDate': 1554513160, 'lastReportWatts': 20, 'maxReportWatts': 232}, {'serialNumber': '000000000013', 'lastReportDate': 1554513164, 'lastReportWatts': 24, 'maxReportWatts': 250}]`

I have looked into adding inverter production collection to your envoy_reader.  Would you want to add this support if I wrote the code?

dalklein, 2019-05-28:
Folks might be interested, I'm sure someone would use it.  Especially if they are not using per panel monitoring from Enlighten, and / or want a completely local only solution.   But the Enlighten per panel monitoring is available and works nice.   In my case (older Envoy, with software before the .json api data was added), I can't get the per panel data directly from the Envoy anyway.

jesserizzo, 2019-06-10:
Just added this, and put in a PR for Home Assistant

closed 2019-06-10 by jesserizzo