jesserizzo/envoy_reader #8: Support for older s/w Envoy C, before s/w R3.9  (no .json page on Envoy)

dalklein, 2019-06-20:
Hello,  thanks for your work making the Envoy work with homeassistant.    

Some older Envoys (mine....)  with software before R3.9, do not have the .json data available, only .html pages, http://envoy/production is the main one.

Based on a month ago old version of envoy_reader, I added support to scrape html page from older Envoy, which does not have the .json based pages available. Working correctly for several weeks with 0.93.1 hassbian installation. I would like to add changes to the current envoy_reader code, if that's all right.

working (based on older version) is here:     https://github.com/dalklein/envoy_reader/

I got the latest version from github and add in my changes, but ....
the latest version of envoy reader uses a new requests_async file which is not in my older ha install.    fails at line 8: import requests_async as requests 

? where can I get this from and use for testing ?  I guess from the main HA git... ?

tried updating to the current ha 0.94.4 figuring it would be included, but that breaks something else (main web interface not working). So I reverted back to 0.93.1 for now.

Any advice welcome,   Thanks,   Dale

dalklein, 2019-06-21:
I believe I have it working as far as I can get it now,  using the previous requests calls,  not the requests_asysc with await.    please see files here,  https://github.com/dalklein/envoy_reader/       and change the requests stuff (not very many places to change),   or please let me know how I need to set up requests_async.        Thanks

jesserizzo, 2019-07-05:
@dalklein Can this issue be closed? I believe you figured it out for that PR I merged a few weeks ago, right?

dalklein, 2019-07-05:
Yes thanks, go ahead and close it.  It is working in my install now.  Looks like you got the changes included in the pr, thanks!

closed 2019-07-05 by dalklein