jesserizzo/envoy_reader #10: "Error while setting up platform enphase_envoy" in HA 0.96.0

DavidDeSloovere, 2019-07-18:
Hi Jesse,
Cross posting this because you're probably not monitoring the home assistant repo.
https://github.com/home-assistant/home-assistant/issues/25224 
Thank you for your time.

dalklein, 2019-07-18:
I left a comment there, hope that helps.  I had the same error using the later envoy code, because I didn't have requests_async installed in the ha venv yet.

gtdiehl, 2019-07-18:
I think that the issue should be re-opened in the Home Assistant repository.  Manually installing `requests-async` is a workaround but the underlying problem should really be fixed, which is to add the module to the `manifest.json` file

DavidDeSloovere, 2019-07-21:
https://github.com/home-assistant/home-assistant/issues/25331 home assistant issue now

closed 2019-07-21 by DavidDeSloovere