jesserizzo/envoy_reader #13: Add support for gathering inverter status

Tim-GitHub, 2019-09-03:
I am using a combination of the envoy_reader and my own Home Assistant value_json template to count the number of inverters based on status. It seems that adding that inverter status into the existing calls may be reasonable, as it is handy for monitoring how many inverters are online.

Here is how I am getting the inventory.json (no password needed):
```
shell_command: 
  get_envoy_inventory: curl "http://internal_ip/inventory.json" > /config/cache/inverter_inventory.json
```

That saves the JSON to my Home Assistant storage, which I then parse with these templates:
```
- platform: file
  name: 'Envoy Inverters Communicating'
  file_path: /config/cache/inverter_inventory.json
  value_template: >
    {% set ns = namespace(total=0) %}
    {% for i in range(17) %}
      {% if value_json[0].devices[i].communicating == true %}
        {% set ns.total = ns.total + 1 %}
      {% endif %}
    {% endfor %}
    {{ns.total}}
  scan_interval: 86400

- platform: file
  name: 'Envoy Inverters Producing'
  file_path: /config/cache/inverter_inventory.json
  value_template: >
    {% set ns = namespace(total=0) %}
    {% for i in range(17) %}
      {% if value_json[0].devices[i].producing == true %}
        {% set ns.total = ns.total + 1 %}
      {% endif %}
    {% endfor %}
    {{ns.total}}
  scan_interval: 86400

- platform: file
  name: 'Envoy Inverters Status OK'
  file_path: /config/cache/inverter_inventory.json
  value_template: >
    {% set ns = namespace(total=0) %}
    {% for i in range(17) %}
      {% if value_json[0].devices[0].device_status[0] == "envoy.global.ok" %}
        {% set ns.total = ns.total + 1 %}
      {% endif %}
    {% endfor %}
    {{ns.total}}
  scan_interval: 86400- platform: file
```
Envoy Inverters Communicating - Counts all inverters that are communicating with the Envoy
Envoy Inverters Producing - Counts all inverters are that producing
Envoy Inverters Status OK - Counts all the inverters that are communicating with the envoy and producing (envoy.global.ok)

In reality, all 3 numbers seem to always match. The inverters only communicate when they have power from the solar panels. When they have power from the solar panels, they are also producing. So only monitoring 'Envoy Inverters Status OK' is likely sufficient. The states that show up in that section of the json are:
```
“envoy.global.ok”
“envoy.cond_flags.pcu_ctrl.commandedreset”
“envoy.cond_flags.pcu_ctrl.dc-pwr-low”
“envoy.cond_flags.obs_strs.discovering”
“envoy.cond_flags.obs_strs.failure”
```
(additional info: https://thecomputerperson.wordpress.com/2016/08/03/enphase-envoy-s-data-scraping/)

So my thinking would be to do something similar, parse the inventory.json and count the number of inverters with a status of "envoy.global.ok", and create a sensor in Home Assistant with the quantity. For my home, all my inverters come online within about 10 minutes of each other, and then remain constant throughout the day. A boring sensor, but mostly this is helpful for identifying when an inverter has issues (offline in the middle of the day).

If this is outside the scope of what you intend with envoy_reader, I fully understand and feel free to close the issue. My current method is working for me, but I thought other users may appreciate the info as well.

jesserizzo, 2019-12-21:
@Tim-GitHub  It already supports the production values from individual inverters based gathered from http://envoy/api/v1/production/inverters (username: 'envoy' password: last 6 digits of the serial number).  Does your suggestion add additional value to that?

jesserizzo, 2020-04-26:
@Tim-GitHub  I'm closing the issue, feel free to reopen and leave a comment clarifying if this would add additional information beyond the existing per inverter production that the sensor supports.

closed 2020-04-26 by jesserizzo