jesserizzo/envoy_reader #14: Envoy Software version D5.0.34 data error

pompompom, 2019-10-31:
Hi,

I have an envoy that was working fine with HA until june. Now I have errors like this 
`"Got` a response from 'http://envoy/production.json', but metric could not be found. Maybe your model of Envoy doesn't support the requested metric."

The command : 
`curl` 'http://envoy/production.json'
Gives :
`{"production":[{"type":"inverters","activeCount":1,"readingTime":1572524639,"wNow":117,"whLifetime":482485}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"w`                                        hNow":0,"state":"idle"}]}

And as the error states, there is not the requested metric for example in the code:  `raw_json["production"][1]["wNow"]` 

There is only 1 element in the production array, and it doesn't contain the wNow attribute.

So it seems that either there is an upgrade of the Json format since a recent enphase upgrade either there is something wrong with my setup.

Is this library valid for this software version ? My model seems to be an S version, it is identified in the code by "PC", maybe the "/api/v1/production" endpoint is the right place to look for the production data now ?

rct, 2019-10-31:
A data point, my Envoy IQ (different/newer? model than the s) is running D5.0.34.  `/production.json` still gives me the two element production array that looks like what others have posted. Element index 0 contains the data from the inverters, Element index 1 has the data from the meters (CTs).

Did anything change w.r.t the metering setup of your envoy?

On my system,  `wattsNow` from `/api/vi/production` gives the same value as `production[0].wNow`.

(I actually had to change a local copy of the code to use `production[0].wNow` because the metering isn't setup correctly on my envoy for a couple of reasons.)

pompompom, 2019-11-04:
Hi, 

I have read somewhere, that the cause may be that I don't have a metered "S" device. And thus the "eim" section containing the production data is not present.
The `/api/v1/production` works correctly. 
Maybe if the content of the `production.json` api request doesn't contain the `eim` section we could rollback to the `api/v1/production` endpoint.

I didn't changed anything to my setup since the intallation, so maybe a recent upgrade triggered this incompatibility.

gtdiehl, 2019-12-19:
I've been writing some unit tests using pytest and ran into this issue.  It appears the current code for retrieving production information is grabbing details from the `eim` section (CT) of json.

I would propose changing
https://github.com/jesserizzo/envoy_reader/blob/a203d357ed520a23ff641670aa85d5a304355187/envoy_reader/envoy_reader.py#L119
to
`production = raw_json["production"][0]["wNow"]`

As there are two versions of the Envoy-S, a standard model that does not have the `eim` section and one that has metering capabilities which has the `eim` section.  Both models will have the `inverters` section in the json output.

gtdiehl, 2019-12-21:
In the PR #20 @jesserizzo mentioned that the eim wNow will update sooner than the inverter wNow.

I'll try to make a change where if the retrieval of the eim wNow fails fallback to the inverter wNow.

gtdiehl, 2019-12-23:
I've updated the production() in PR #20 in the envoy_reader library to retrieve wNow data from the inverters using `production = raw_json["production"][0]["wNow"]` if the eim wNow data from `production = raw_json["production"][1]["wNow"]` is not available

https://github.com/jesserizzo/envoy_reader/blob/fa986d10696cb74b76ee31cce45d1a4f1ec11445/envoy_reader/envoy_reader.py#L119-L122

commit 9ad3d532 (2019-12-25): Change production retrieval

Since there are two different Envoy-S models, one that has only inverters and one with inverters and CTs (eim).  The production method should retrieve watts from the inverters as all Envoy-S will have inverters.  Closes #14
closed 2019-12-25 by jesserizzo