jesserizzo/envoy_reader #19: Add testing framework

gtdiehl, 2019-12-19:
I am investigating a testing framework to do testing when changes are made to the envoy_reader module.

Currently I am writing tests using pytest.  Not having any experience with testing frameworks is pytest okay?  If so I can open a PR to merge in the tests I have created so far.

jesserizzo, 2019-12-21:
I don't have any opinion regarding python testing frameworks, if that seems good then go for it.

dennisdegreef, 2019-12-30:
If you want any help with this, I'd gladly help out. I'm not fluent with python perse, but I am with testing frameworks and methodologies from other languages.

gtdiehl, 2019-12-30:
@dennisdegreef Great!  I'll open a PR to merge in what I have created so far.

gtdiehl, 2021-04-16:
Closing issue.  We have a testing framework.

closed 2021-04-16 by gtdiehl