jesserizzo/envoy_reader #36: Add storage data

dutchkiwi2, 2020-07-20:
If the Envoy is a PC type, it can support storage devices.
Can a check if storage data is available be included and storage details added.

Having data available on the charge / discharge (wNOW) and percentage full (percentFull) are very usefull in HomeAssist.

The production.json files includes the following data when storage is attached:

"storage": [
    {
      "type": "acb",
      "activeCount": 4,
      "readingTime": 1595203104,
      "wNow": -178,
      "whNow": 285,
      "state": "charging",
      "percentFull": 6
    }

gtdiehl, 2020-11-04:
@dutchkiwi2 are these storage devices the Encharge 3/10 storage systems?

dutchkiwi2, 2020-11-04:
The example is from acb batteries, the predecessor of encharge

gtdiehl, 2020-11-04:
Ok great I found the data sheets for that product.

One other question, do you have 4 acb units installed?

dutchkiwi2, 2020-11-04:
Yes we do.

bouyssic, 2020-12-13:
Hi team. 

I would be interested on this feature too. If you need help with testing let me know.

Cheers

gtdiehl, 2021-04-22:
@bouyssic Do you have the ACB or the Encharge 3/10 units?

dutchkiwi2, 2021-04-22:
We have ACB batteries

Johan & Monique
sent from my phone

gtdiehl, 2021-04-23:
I'm working on adding battery data to the ```envoy_reader``` API have to test it on other firmwares and such.

Also working on adding it to Home Assistant, this is what I have so far on that side:
[image]

gtdiehl, 2021-04-23:
I just noticed that @dutchkiwi2 your JSON output contains ```percentFull```. I need to add that attribute. It appears this only appears when you have batteries installed, but I can't confirm this as yours is the only Envoy I've ever seen that has batteries installed

bouyssic, 2021-04-23:
Hello @gtdiehl ,

I have ACB also like @dutchkiwi2 

Here an example:
`    "storage": [
        {
            "type": "acb",
            "activeCount": 1,
            "readingTime": 1619161595,
            "wNow": -3,
            "whNow": 0,
            "state": "idle",
            "percentFull": 0
        }`

If you need anything for debugging I can help. Right now I deleted the integration and am gathering all information through Restful Sensors

gtdiehl, 2021-04-23:
@bouyssic I have made my changes to the code to retrieve the battery status. Can you run the ```envoy_reader.py``` standalone code against your Envoy? The code is located at [envoy_reader.py](https://raw.githubusercontent.com/gtdiehl/envoy_reader/battery_status/envoy_reader/envoy_reader.py)

bouyssic, 2021-04-23:
Here you go sir: 

`bash-5.0# python envoy_reader.py 
Enter the Envoy IP address or host name, or press enter to use 'envoy' as default: 1.1.1.1
Enter the Username for Inverter data authentication, or press enter to use 'envoy' as default: username
Enter the Password for Inverter data authentication, or press enter to use the default password: ******
Reading...
production:              1237
consumption:             558
daily_production:        1286
daily_consumption:       15771
seven_days_production:   108580
seven_days_consumption:  274603
lifetime_production:     3182116
lifetime_consumption:    12788787
inverters_production:    {'122003050569': [118, '2021-04-23 10:36:47'], '122003037804': [121, '2021-04-23 10:36:55'], '122003031864': [117, '2021-04-23 10:36:44'], '122003050565': [121, '2021-04-23 10:36:51'], '122003052183': [122, '2021-04-23 10:36:58'], '122003038059': [117, '2021-04-23 10:37:01'], '121944079829': [121, '2021-04-23 10:37:03'], '122003050079': [116, '2021-04-23 10:36:44'], '122003032003': [117, '2021-04-23 10:36:52'], '122003031789': [115, '2021-04-23 10:36:51'], '122003050570': [119, '2021-04-23 10:36:57']}
battery:                 {'type': 'acb', 'activeCount': 1, 'readingTime': 1619167023, 'wNow': 121, 'whNow': 74, 'state': 'discharging', 'percentFull': 6}
`

gtdiehl, 2021-04-23:
Great! Thank you. I will merge these changes and make a new release. Than I can work on getting the Home Assistant sensor updated.

bouyssic, 2021-04-23:
Don't thank me, you did all the heavy lifting

closed 2021-04-23 by gtdiehl

gtdiehl, 2021-04-24:
@bouyssic I have to wait for Home Assistant release that contains the Enphase Envoy ```config_flow UI``` change, [PR #48517](https://github.com/home-assistant/core/pull/48517). I'm thinking it will be in the 2021.5 release. I'll reach out once you can test the updated sensor code, that is if you want to upgrade.

Unless you have a Home Assistant development environment already setup?

PR #74 (2021-04-23): Add Battery storage data
Adding Battery storage data that is coming from ```/production.json```

```
"storage": [{"type": "acb","activeCount": 4,"readingTime": 1595203104,"wNow": -178,"whNow": 285,"state": "charging","percentFull": 6}
```

Battery storage data will only be returned when the battery is installed in the Envoy system. To determine if the battery is installed we check for the presence of the ```percentFull``` attribute, as in the above example.

If a battery is not present the ```/production.json``` page will return:
```
"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wN [... cut]