jesserizzo/envoy_reader #46: Support reading values of consumption and production per phase

pedrocr, 2020-11-24:
As discussed in #37 when using a three-phase system there is a lot of other data the equipment captures that is not being read. Apparently with installer permissions there's an API that gives things like total power, reactive power, frequency, voltage for each of the three phases in `/stream/meter`.

There's also this reverse engineering to generate the needed password from the serial number:

https://thecomputerperson.wordpress.com/2016/08/28/reverse-engineering-the-enphase-installer-toolkit/

So maybe it would be possible to grab this API end-point as well and get some very rich data. The stream is even apparently around once per second in a continuously open connection so it should create a great power meter info.

pedrocr, 2020-11-24:
Apparently there's also a `/ivp/meters/readings` that should include the three phases separately.

lnlp, 2020-11-25:
The `/stream/meter` output is generated about every second.
My assumption is that this was implemented for setup/installation purposes.

I wonder though, if this could negatively impact the performance of the Envoy in its communication with the inverters.
It should not, but this depends on if the hardware on the Envoy is sufficiently powerful to do this on regular/continuous bases.

My (Dutch) smart power meter reports data every single second. I would love to see the same from the Envoy.
While inverter data is updated only every 5 minutes, the update frequency of data from the current transformers may be much higher though (maybe even once a second). I am unable to verify this however because I have no CT's installed.

pedrocr, 2020-11-25:
Since the output is a stream where we only get values when the meter decides instead of continuously hitting it with requests it should theoretically be less of a performance drag. It doesn't have to continuously react to new connections and only outputs values at its own rate. Enphase themselves tell me this per-phase information is available for premium subscribers so they are already expecting this to work continuously and not just for installers.

pedrocr, 2020-11-25:
Apparently there's an URL without authentication that's confirmed to give per-phase values:

https://github.com/jesserizzo/envoy_reader/issues/37#issuecomment-733861030

lnlp, 2020-11-25:
And another (familiar) one that can also give per-phase values:

See the following [comment in issue #37](https://github.com/jesserizzo/envoy_reader/issues/37#issuecomment-733962925)

OllemGit, 2020-11-26:
Sample output with just consumption CT's (3 phases) installed:

{
  "production": [
    {
      "type": "inverters",
      "activeCount": 14,
      "readingTime": 1606391880,
      "wNow": 135,
      "whLifetime": 1491812
    },
    {
      "type": "eim",
      "activeCount": 0,
      "measurementType": "production",
      "readingTime": 1606392132,
      "wNow": 0.919,
      "whLifetime": 0.982,
      "varhLeadLifetime": 1.333,
      "varhLagLifetime": 3.617,
      "vahLifetime": 597.594,
      "rmsCurrent": 0.315,
      "rmsVoltage": 698.098,
      "reactPwr": 0.489,
      "apprntPwr": 73.331,
      "pwrFactor": 0,
      "whToday": 0.982,
      "whLastSevenDays": 0.982,
      "vahToday": 3.594,
      "varhLeadToday": 0.333,
      "varhLagToday": 0.617,
      "lines": [
        {
          "wNow": 0,
          "whLifetime": 0.001,
          "varhLeadLifetime": 0.102,
          "varhLagLifetime": 0.054,
          "vahLifetime": 9.395,
          "rmsCurrent": 0,
          "rmsVoltage": 232.931,
          "reactPwr": 0,
          "apprntPwr": 0,
          "pwrFactor": 0,
          "whToday": 0,
          "whLastSevenDays": 0,
          "vahToday": 0,
          "varhLeadToday": 0,
          "varhLagToday": 0
        },
        {
          "wNow": 0.919,
          "whLifetime": 0.979,
          "varhLeadLifetime": 0.021,
          "varhLagLifetime": 3.563,
          "vahLifetime": 538.178,
          "rmsCurrent": 0.315,
          "rmsVoltage": 232.885,
          "reactPwr": 0.489,
          "apprntPwr": 73.331,
          "pwrFactor": 0,
          "whToday": 0,
          "whLastSevenDays": 0,
          "vahToday": 0,
          "varhLeadToday": 0,
          "varhLagToday": 0
        },
        {
          "wNow": 0,
          "whLifetime": 0.002,
          "varhLeadLifetime": 1.21,
[... 152 lines cut ...]
          "rmsCurrent": 1.817,
          "rmsVoltage": 232.352,
          "reactPwr": -317.683,
          "apprntPwr": 422.104,
          "pwrFactor": 0.4,
          "whToday": 0,
          "whLastSevenDays": 0,
          "vahToday": 0,
          "varhLeadToday": 0,
          "varhLagToday": 0
        }
      ]
    }
  ],
  "storage": [
    {
      "type": "acb",
      "activeCount": 0,
      "readingTime": 0,
      "wNow": 0,
      "whNow": 0,
      "state": "idle"
    }
  ]
}

pedrocr, 2020-11-26:
It seems the only data this may not include is frequency, although I'm not sure that's actually being measured or if it's just a fixed value. `/stream/meter` may still be better as a way to get more frequent data to react to power variations. But for reporting purposes this seems like the best solution.

lnlp, 2020-11-26:
@OllemGit 
How are your consumption CT's wired?

One of the wNow values is negative. I assume this means feeding back to the grid.

This means that the wires you measure do not only carry consumption flow but also any production that flows back to the grid.

Another way to wire the CT's is to measure only consumption, that way you measure true consumption (regardless of wheter the power came from the grid or the solar panels).

lnlp, 2020-11-26:
> It seems the only data this may not include is frequency

I wonder if frequency is any interesting. The local net here is 50 Hz based.

In the Envoy I see the following values defined:

```
      "frequency_limits": {
        "label": "Frequency limits",
        "parameters": {
          "fast_over_frequency_trip_time": {
            "label": "Over frequency trip time",
            "parameter_type": "tpm",
            "display_value": "2,000",
            "units": "ms",
            "table": "Over frequency disconnect",
            "row": "Limit",
            "column": "Time"
          },
          "fast_over_frequency_limit": {
            "label": "Over frequency limit",
            "parameter_type": "tpm",
            "display_value": "51.00",
            "units": "Hz",
            "table": "Over frequency disconnect",
            "row": "Limit",
            "column": "Magnitude"
          },
          "fast_under_frequency_trip_time": {
            "label": "Under frequency trip time",
            "parameter_type": "tpm",
            "display_value": "2,000",
            "units": "ms",
            "table": "Under frequency disconnect",
            "row": "Limit",
            "column": "Time"
          },
          "fast_under_frequency": {
            "label": "Under frequency limit",
            "parameter_type": "tpm",
            "display_value": "48.00",
            "units": "Hz",
            "table": "Under frequency disconnect",
            "row": "Limit",
            "column": "Magnitude"
          },
          "reconnect_ac_frequency_high": {
            "label": "Reconnect AC frequency high",
            "parameter_type": "tpm",
            "display_value": "50.05",
            "units": "Hz",
            "table": "Reconnect frequency",
            "row": "High",
            "column": "Magnitude"
          },
          "reconnect_ac_frequency_low": {
            "label": "Reconnect AC frequency low",
            "parameter_type": "tpm",
            "display_value": "48.00",
            "units": "Hz",
            "table": "Reconnect frequency",
            "row": "Low",
            "column": "Magnitude"
          }
        }
      },
```

So the total margin (of operation) is quite small: only 4 Hz.

OllemGit, 2020-11-26:
> @OllemGit
> How are your consumption CT's wired? Apparently as "Load with solar production".
> 
> One of your consumption _wNow_ values above is negative, which would mean 'production feeding back to the grid'.
> 
> There are two ways to **_wire_** (and accordingly manually _configure_) consumption CT's:
> 
>     * "Load with solar production"
> 
>     * "Load only"
> 
> 
> The first measures the actual value of `consumption - production`.
> The second measures the actual true `consumption` values.
> 
> I wonder how the Envoy handles both situations and how it calculates net-consumption and total-consuption values.
> 
> 
> _I would expect that the Envoy knows which inverter is connected to which phase (based on doing an inventory). But I have been unable to find a relation between inverter and phase/line in the available data._

The fact I only have consumption CT's and no production CT's is not my choice, but the installation firm initially didn't want to install any CT's at all. Because the Envoy already measures production we agreed to install only the consumption ones.

They selected 'Load Only' in the configuration.

I haven't been able to figure out if there are any bogus values measured.
To make things more complicated I have some more solar panels which are using a regular inverter (Onmik). 
(The production ratio is around 2:3 Omnik:Enphase 
So this might also be the reason for showing some strange numbers?

lnlp, 2020-11-27:
I currently have no CT's installed, so everything below will be a theoretical excercise based on Envoy documentation, my Envoy's firmware (Web UI/API) and experience.

_Be aware that the information below contains some (common sense) assumptions._

In my Envoy-S Installer setup menu two options are possible for configuring the Consumption CT's:
- "Load only"
- "Load with solar production"

This is not just a configuration setting but very probably needs to correspond to how the CT are physically wired.
I have tried to visualize both options in below image.

In the 'Envoy-S Metered Multiphase Installation and Operation' manual only a single wiring option is documented,
while the firmware provides the two different options listed above.
The wiring documented in the manual corresponds to a "Load with solar production" situation because both consumption and production current flows through the metered wires.

To create a "Load only" situation the Consumption CT's will have to be wired as depicted in green on the image. 
Only then will the CT's measure only Consumption and no Production.
Although this option is not further described in the manual, it _can_ be selected in the firmware.

[image]

In the "Load only" case the CT's will only measure true actual consumption, and because the production plays no part here the measured values are/should be only positive and never negative.

In the "Load with solar production" case the CT's will only measure the _difference_ between consumption and production (per phase). When production on a phase is 100W and consumption on that phase is 200W, 100W will flow from the solar panels to the consumer and the other 100W will be coming from the grid: the consumption CT for that phase will measure `consumption - production` in this case 200W - 100W = 100W.
When production is higher than consumption, e.g. production is 300W and consumption is 200W then the consumption CT should measure 200W - 300W = -100W.

@OllemGit This _could_ explain why we see a negative value in your consumption values.
So possibly your CT's are wired according to "Load with solar production" instead of "Load only".

The problem with "Load with solar production" is that production must also be metered in order to calculate the true consumption values. But in your situation the production CT's are not installed.

The Envoy could possible use production data from the inverters as substitute for the missing CT metered production. But that is only updated once every 5 minutes but the results probably would not be accurate.
I also wonder whether the Envoys actually knows which micro-inverters are connected to which phase (this would be required to use production data from the inverters as substitue for missing CT production data).
I suspect however, that the Envoy will not use the inverter data as substitute.

My feeling is that without production CT's installed. Wiring the consumption CT's according to "Load only" will provide the most reliable consumption data.

I'm interested in what data is actually used for Enpase Enlighten in the cloud.

@OllemGit Do you see consumption data/graphs in Enlighten and is that data reliable?

OllemGit, 2020-11-27:
Thanks a lot for sharing. I'll open the covers somewhere this weekend to check the wiring!

HA-TB303, 2020-12-08:
> I currently have no CT's installed, so everything below will be a theoretical excercise based on Envoy documentation, my Envoy's firmware (Web UI/API) and experience.
> 
> _Be aware that the information below contains some (common sense) assumptions._
> 
> In my Envoy-S Installer setup menu two options can be selected for configuring the Consumption CT's:
> (visible in manual metering setup)
> 
> * "Load only"
> * "Load with solar production"
> 
> This is not just a configuration setting but very probably needs to correspond to how the CT's are physically wired.
> I have tried to visualize both options in below image.
> 
> In the 'Envoy-S Metered Multiphase Installation and Operation' manual only a single wiring option is documented,
> while the firmware provides the two different options listed above.
> The wiring documented in the manual corresponds to a "Load with solar production" situation because both consumption and production current can flow through the CT metered wires (actually their delta and the direction is determined by which of both is higher).
> 
> To create a "Load only" situation the Consumption CT's will have to be wired as depicted in green on the image.
> Only then will the CT's measure pure total-consumption.
> While this option is not described in the manual, it _can_ be selected in the firmware.
> _(The manual is not exhaustive and lacks certain details and explanations.)_
> 
> An advantage of "Load with solar production" wiring is that _all_ consumption is measured, including power used by the Q-Relay and the Envoy. It is yet unclear however if this will also work reliable when only consumption CT's installed.
> 
> An advantage of "Load only" wiring may be, in case only consumption CT's are installed (and "Load with solar production" can work only when production CT's are installed), that you still get actual total-consumption measurements. And when using `stream/meter` to read them that they can be combined with data reported by a smart electricity meter (e.g. every second by a Dutch DSMR5 compliant smart meter) for use in Home Assistant, Grafana etc.. The disadvantage is that this will not measure power usage from the Envoy itself and the Q-Relay (so your own consumption measurements will not 100% exactly match the consumption measurement by your electricity meter and corresponding energy bill).
> 
> _Maybe others have experience with "Load only" wiring of their consumption CT's and can share their experience here._
> 
> [image]
> 
> _Note that I enhanced an existing image. The black consumption CT's could also be placed **after** the primary circuit breaker and the green consumption CT"s could also be placed **before** the residual current circuit breakers. Whichever is most practical to implement in existing installations._
> 
> In the "Load only" case the CT's will only measure true actual consumption, and because the production plays no part here the measured values are/should be only positive and never negative.
> 
> In the "Load with solar production" case the CT's will measure the _difference_ between consumption and production (per phase). When production on a phase is 100W and consumption on that phase is 200W, 100W will flow from the solar panels to the consumer and the other 100W will be coming from the grid: the consumption CT for that phase will measure `consumption - production` in this case 200W - 100W = 100W.
> When production is higher than consumption, e.g. production is 300W and consumption is 200W then the consumption CT should measure 200W - 300W = -100W (which is unused production flowing back to the grid).
> 
> @OllemGit This _could_ explain why there is a negative value in your consumption values.
> So possibly your CT's are wired according to "Load with solar production" instead of "Load only".
> (Other possible causes could be incorrect direction placement of a CT or its wires being switched).
> 
> The problem with "Load with solar production" is that production must also be metered in order to calculate the true consumption values. But in your situation the production CT's are not installed.
> 
> The Envoy could possibly use production data from the inverters as substitute for the missing CT metered production. But that is only updated once every 5 minutes but the results probably would not be accurate.
> I also wonder whether the Envoy actually knows which micro-inverters are connected to which phase (this would be required to use production data from the inverters as substitute for missing CT production data).
> I suspect however, that the Envoy will not use the inverter data as substitute.
> 
> _In worst case I could be totally mistaking and "Load with solar production" could possibly mean 'consumption with production CT's also installed' but I guess not because the firmware will already know if production CT's are enabled (in the configuration)._
> 
> **Interesting questions:**
> 
> * Is an Envoy installation with only consumption CT's installed actually a supported configuration?
>   I guess so because in manual metering configuration (in installer setup) one can have zero production CT's selected while selecting 1, 2 or 3 consumption CT's.
> * If installing only consumption CT's without installing production CT's is supported, is it still possible to wire them as "Load with solar production" or would this require them to be wired as "Load only" (because live CT production data is missing to calculate total-consumption)?
[... cut ...]
               "rmsVoltage":231.655,
               "reactPwr":-247.068,
               "apprntPwr":380.231,
               "pwrFactor":0.54,
               "whToday":0,
               "whLastSevenDays":0,
               "vahToday":0,
               "varhLeadToday":0,
               "varhLagToday":0
            }
         ]
      }
   ],
   "storage":[
      {
         "type":"acb",
         "activeCount":0,
         "readingTime":0,
         "wNow":0,
         "whNow":0,
         "state":"idle"
      }
   ]
}
```

HA-TB303, 2020-12-08:
For now I am getting phase data "the hacky way" like outlined here:
https://community.home-assistant.io/t/need-testers-for-updated-enphase-envoy-sensor/213296/51?u=michel72

which now looks like:
[image]

lnlp, 2020-12-15:
@HA-TB303 

> I have exactly this setup:

The image actually shows 2 different setups:

1. Consumption CT's drawn in black: "Load with solar production"
2. Consumption CT's drawn in green: "Load only"

How are your Consumption CT's setup?

HA-TB303, 2020-12-15:
Ah, I missed that. They are on the grid side (black in the image)

HA-TB303, 2020-12-15:
I have checked again. Mine are positioned between the residual current circuit breakers and the main 3 phase circuit breaker.

HA-TB303, 2020-12-16:
> @HA-TB303
> 
> > I have exactly this setup:
> 
> The image actually shows 2 different setups:
> 
> 1. Consumption CT's drawn in black: "Load with solar production"
> 2. Consumption CT's drawn in green: "Load only"
> 
> How are your Consumption CT's setup?

Regarding this, is there actually a different setup required in the Enphase Envoy-S Metered, depending of the choice one makes for installing the consumption CT's, if so, where to find this setting?

I would just like to verify if it's setup correctly by the installer.

testuser7, 2022-06-02:
Any news with this issue? Are you planning to add support for the `/production.json?details=1` endpoint?

HA-TB303, 2022-12-15:
Yeah, that would be great! I would like to create automations that use produced solar energy per phase and my ' hacky'  way won't work anymore since v7 :(

DanielNagy, 2023-06-26:
+1 for support for the /production.json?details=1 endpoint. Would love to see additional 3phase level data in Home assistant.

testuser7, 2023-07-07:
> +1 for support for the /production.json?details=1 endpoint. Would love to see additional 3phase level data in Home assistant.

You can use this HACS integration: https://github.com/posixx/home_assistant_custom_envoy