jesserizzo/envoy_reader #62: Questions about negative whattsNow value

lnlp, 2021-01-22:
Today I noticed reporting of negative `whattsNow` value in the Home Assistant Enphase Envoy plugin (labeled 'Actueel' on screen):

[image]

However `envoy/api/v1/production` showed an active `whatssNow` value of 0:

[image]

A short moment later (I refreshed both screens) the values were as follows:

[image]

[image]

I'm not sure why the Envoy does report negative whattsNow values and this is not something caused by `envoy_reader`.

It is strange however that the negative whattsNow values reported in Home Assistant are different from the ones reported by the Envoy api `envoy/api/v1/production`.

This is either caused by a problem in envoy_reader or the Home Assistant Enphase Envoy plugin.

How come that the Envoy api reports a negative `whattsNow` value?

_Values were observed at end of noon while it was almost dark. Envoy Metering Multiphase without CT metering installed._

dalklein, 2021-01-22:
'end of noon'?   If it's dark out, the micros do consume a small amount of current.  

In my system with 23 total inverters and the older oval Envoy,  m190 / m210 and replacement m190IG/IQ7,  the system does consume 10 to 11 watts total at night, measured by my utility meter.  My Envoy is powered directly from the same circuit so that would be using 1-3watts.    I see zero or some small value reported by the Envoy at night, 10 to 40 watts,  and I think this may be the last values reported from the micros before they turn off.

lnlp, 2021-01-22:
Thanks. It now shows 0 again (which I would expect). I apparently looked exactly at the 'right' time to notice the effect.

The micro inverters use power overnight? I have 15 iQ7+ installed. Maybe the iQ7(+) make a difference in that aspect?

Still strange that the value reported in Home Assistant was different from the value reported by the API.

gtdiehl, 2021-04-15:
@lnlp Sorry for the late reply. If I remember correctly you have an Envoy-S? The current watts now value would be coming from ```/production.json```. This could account for the difference.