jesserizzo/envoy_reader #72: Add Envoy data from Encharge batteries

rsnodgrass, 2021-04-17:
## Use Case

The Envoy now supports collecting data from the Enpower transfer switch and any (0 or more) Encharge batteries attached.

## Solution

Sensor data for battery consumption, battery charging, current charge, and like the existing individual microinverter functionality the ability to have current charge for each independent battery.

## Alternatives

rsnodgrass, 2021-04-18:
I see that accessing the /home URL locally shows that Enpower device is installed and that the Grid is ON. 

Some other data end points locally that return data on the setup:

```
http://{}/ivp/meters
[
    {
        "eid": 3453453,
        "state": "enabled",
        "measurementType": "production",
        "phaseMode": "split",
        "phaseCount": 2,
        "meteringStatus": "normal",
        "statusFlags": []
    },
    {
        "eid": 124124124,
        "state": "enabled",
        "measurementType": "net-consumption",
        "phaseMode": "split",
        "phaseCount": 2,
        "meteringStatus": "normal",
        "statusFlags": []
    }
]
```

rsnodgrass, 2021-04-18:
And..importantly the following has data on the Enpower device:

"enpower":{"connected":true,"grid_status":"closed"}

I believe grid_status closed = that the grid is connected (e.g. relay is closed, not in an open state).

```
http://{}/home.json

{"software_build_epoch":1610420483,"is_nonvoy":false,"db_size":17600512,"db_percent_full":"  4.66","timezone":"US/Pacific","current_date":"04/17/2021","current_time":"20:24","network":{"web_comm":true,"ever_reported_to_enlighten":true,"last_enlighten_report_time":1618715554,"primary_interface":"wlan0","interfaces":[{"type":"ethernet","interface":"eth0","mac":"00:1D:C0:68:BA:BC","dhcp":true,"ip":"169.254.120.1","signal_strength":0,"signal_strength_max":1,"carrier":false},{"signal_strength":3,"signal_strength_max":5,"type":"wifi","interface":"wlan0","mac":"50:F1:4A:EC:5B:58","dhcp":true,"ip":"192.168.50.22","carrier":true,"supported":true,"present":true,"configured":true,"status":"connected"}]},"tariff":"single_rate","comm":{"num":0,"level":0,"pcu":{"num":0,"level":0},"acb":{"num":0,"level":0},"nsrb":{"num":0,"level":0},"encharge":[{"num":0,"level":0,"level_24g":0,"level_subg":0}]},"alerts":[],"update_status":"satisfied","wireless_connection":[{"signal_strength":0,"signal_strength_max":0,"type":"zigbee","connected":true},{"signal_strength":0,"signal_strength_max":0,"type":"subghz","connected":true}],"enpower":{"connected":true,"grid_status":"closed"}}
```

gtdiehl, 2021-04-22:
@rsnodgrass I don't have access to an Envoy currently with batteries, are those URLs password protected?

rsnodgrass, 2021-04-22:
@gtdiehl sent you an email with more info.

gtdiehl, 2021-04-29:
I've looked into this a little bit more after implementing the Battery status for #36. 

It seems adding the grid connection status of the Enpower Smart Switch might be beneficial to people as seems to show if you are connected to the grid or not.
```
"enpower":{
  "connected":true,
  "grid_status":"closed"
}
``` 

This would require polling of the ```/home.json``` page. If we do poll this page we should probably investigate what the other values in the ```comm``` section mean.

At this point I can look into retrieving and displaying the Enpower Smart Switch ```connected``` and ```grid_status``` values. 

I can implement it to check for the presence of the ```connected``` key, if the key is present display the status of the Encharge device and the grid status. And if the key is not present the API will return that the Encharge is not currently supported with the Envoy.

cgarwood, 2021-05-19:
I noticed my Encharge batteries don't show up under the `production.json` output. I did some digging, and it looks like my envoy shows the Encharge and Enpower data under `/inventory`

```
[
    {
        "type": "ENCHARGE",
        "devices": [
            {
                "part_num": "830-00703-r67",
                "installed": 1621354342,
                "serial_num": "xxx",
                "device_status": [
                    "envoy.global.ok",
                    "prop.done"
                ],
                "last_rpt_date": 1621464675,
                "admin_state": 6,
                "admin_state_str": "ENCHG_STATE_READY",
                "created_date": 1621354342,
                "img_load_date": 1621354342,
                "img_pnum_running": "1.2.2883_release/20.34",
                "zigbee_dongle_fw_version": "0x1009",
                "operating": true,
                "communicating": true,
                "sleep_enabled": false,
                "percentFull": 98,
                "temperature": 25,
                "maxCellTemp": 26,
                "comm_level_sub_ghz": 5,
                "comm_level_2_4_ghz": 5,
                "led_status": 13,
                "real_power_w": 216,
                "dc_switch_off": false,
                "encharge_rev": 1,
                "encharge_capacity": 3300
            },
            {
                "part_num": "830-00703-r67",
                "installed": 1621354473,
                "serial_num": "xxx",
                "device_status": [
                    "envoy.global.ok",
                    "prop.done"
                ],
                "last_rpt_date": 1621464696,
                "admin_state": 6,
                "admin_state_str": "ENCHG_STATE_READY",
                "created_date": 1621354473,
                "img_load_date": 1621354473,
                "img_pnum_running": "1.2.2883_release/20.34",
                "zigbee_dongle_fw_version": "0x1009",
                "operating": true,
                "communicating": true,
                "sleep_enabled": false,
                "percentFull": 97,
                "temperature": 25,
                "maxCellTemp": 26,
                "comm_level_sub_ghz": 5,
                "comm_level_2_4_ghz": 5,
                "led_status": 13,
                "real_power_w": 317,
                "dc_switch_off": false,
                "encharge_rev": 1,
                "encharge_capacity": 3300
            },
            {
                "part_num": "830-00703-r67",
                "installed": 1621354587,
                "serial_num": "xxx",
                "device_status": [
                    "envoy.global.ok",
                    "prop.done"
                ],
                "last_rpt_date": 1621464802,
                "admin_state": 6,
                "admin_state_str": "ENCHG_STATE_READY",
                "created_date": 1621354587,
                "img_load_date": 1621354587,
                "img_pnum_running": "1.2.2883_release/20.34",
                "zigbee_dongle_fw_version": "0x1009",
                "operating": true,
                "communicating": true,
                "sleep_enabled": false,
                "percentFull": 97,
                "temperature": 25,
                "maxCellTemp": 25,
                "comm_level_sub_ghz": 5,
                "comm_level_2_4_ghz": 5,
                "led_status": 13,
                "real_power_w": -516,
                "dc_switch_off": false,
                "encharge_rev": 1,
                "encharge_capacity": 3300
            }
        ]
    },
    {
        "type": "ENPOWER",
        "devices": [
            {
                "part_num": "860-00276-r28",
                "installed": 1621354111,
                "serial_num": "xxx",
                "device_status": [
                    "envoy.global.ok",
                    "prop.done"
                ],
                "last_rpt_date": 1621464851,
                "admin_state": 24,
                "admin_state_str": "ENPWR_STATE_OPER_CLOSED",
                "created_date": 1621354111,
                "img_load_date": 1621354111,
                "img_pnum_running": "1.2.2064_release/20.34",
                "zigbee_dongle_fw_version": "0x1009",
                "operating": true,
                "communicating": true,
                "temperature": 79,
                "comm_level_sub_ghz": 5,
                "comm_level_2_4_ghz": 5,
                "mains_admin_state": "closed",
                "mains_oper_state": "closed",
                "Enpwr_grid_mode": "multimode-ongrid",
                "Enchg_grid_mode": "multimode-ongrid",
                "Enpwr_relay_state_bm": 496,
                "Enpwr_curr_state_id": 16
            }
        ]
    },
    {
        "grid_profile_name": "IEEE 1547 default 2015",
        "id": "91937832-159a-410a-9594-0a964372e096:0",
        "grid_profile_version": "1.0.11",
        "item_count": 2997
    }
]
```

cgarwood, 2021-05-20:
FWIW to follow-up on this, the `real_power_w` from the `ivp/ensemble/inventory` page seems to differ from what shows up on Enlighten, My batteries show about 0.5kW higher from the envoy inventory data than what Enlighten shows.

thetravellor, 2021-10-05:
I cant offer support for an enhancement that is constrained only to Encharge batteries, add the older and more common ACB batteries and Ill happily help out.

gtdiehl, 2022-01-18:
With the added `envoy_utils` library the `envoy_reader` can now determine the password for the `installer` account. I can see about working on this Issue after #86 is resolved.

jrutski, 2022-07-27:
Interestingly, my /ivp/ensemble/inventory does not show real_power_w for my encharges.

`            {
                "part_num": "836-00750-r25",
                "installed": 1658765836,
                "serial_num": "492203003343",
                "device_status": [
                    "envoy.global.ok",
                    "prop.done"
                ],
                "last_rpt_date": 1658929302,
                "admin_state": 6,
                "admin_state_str": "ENCHG_STATE_READY",
                "created_date": 1658765836,
                "img_load_date": 1658765836,
                "img_pnum_running": "2.0.4396_rel/21.19",
                "zigbee_dongle_fw_version": "100D",
                "bmu_fw_version": "2.1.16",
                "operating": true,
                "communicating": true,
                "sleep_enabled": false,
                "percentFull": 45,
                "temperature": 31,
                "maxCellTemp": 31,
                "comm_level_sub_ghz": 2,
                "comm_level_2_4_ghz": 2,
                "led_status": 12,
                "dc_switch_off": false,
                "encharge_rev": 255,
                "encharge_capacity": 3500
            },`

It appears to be pulling from the IQ8X-BAT micros, which are not exposed anywhere that I can find.