jesserizzo/envoy_reader #78: Enphase integration not working after Envoy updated to D7.0.43 (8ed8ea)

mikeykau, 2021-06-18:
Authentication process appears to have changed.

enkrypt3d, 2021-12-28:
Following this issue.

gtdiehl, 2021-12-28:
After quickly looking at some threads it appears the user creates a token from the [Enphase](http://entrez.enphaseenergy.com/) site.

Here is some [discussion](https://community.home-assistant.io/t/enhpase-envoy-on-2021-8-with-new-energy-feature/328668?page=18) from the Home Assistant forum.

I don't have access to an Enphase Envoy running D7.x firmware. Even when I access the [Enphase](http://entrez.enphaseenergy.com/) site my system does not appear under commissioned envoy. I can only create a token for an uncommissioned envoy. In my test, the JWT payload contains an expiration date of 12 hours after the creation of the token. Not sure about commissioned tokens, as there is some confusion in the forum. A commissioned token had an exp of 7 days, while another was 1 hour.

In any case, even with a token that lasts at most 7 days, the envoy_reader code would need to be enhanced to retrieve a token programmatically. This [Issue](https://github.com/dlmcpaul/EnphaseCollector/issues/22#issuecomment-973656321) has some code in Go that could be ported over to Python.

For me, this is really not possible until I get access to an Envoy running D7.x firmware.

enkrypt3d, 2021-12-28:
There is the token authentication as well as the SSO which works if I login
directly to the envoy. The token configuration is a bit of a mystery but it
does not appear to be required.

gtdiehl, 2021-12-28:
What login do you use for SSO? Is it the Envoy local account like admin or installer? Or is it your login for Enphase Enlighten?

From what I'm reading in all of the forums, is that to access the backend Envoy JSON data a token is needed.

enkrypt3d, 2021-12-28:
It is the same login for enlighten yes.

gtdiehl, 2021-12-28:
Unfortunately for me even if I had access to another's Envoy system, my Enlighten credentials would probably not work on their system. If the backend Envoy JSON data is available without a token from the [Enphase](https://entrez.enphaseenergy.com/) site and only the Enlighten credentials tied to that system are needed, someone with the system and credentials would have to do some investigation and hack some code together to get the ball rolling.

enkrypt3d, 2022-01-01:
Let me how I can assist in contributing.

gtdiehl, 2022-01-08:
@enkrypt3d I was looking at the Go code in my previous post and seeing how that can be ported over somehow. But you mentioned that SSO is possible, though all of the screenshots (as I don't have access to an Envoy running D7.x firmware) have the Token text field entry.

I assume this is what happens when going to ```http://x.x.x.x/production.json```?
[image]

I was wondering can you post a screenshot showing the SSO? And how it is accessed?

enkrypt3d, 2022-01-08:
> @enkrypt3d I was looking at the Go code in my previous post and seeing how that can be ported over somehow. But you mentioned that SSO is possible, though all of the screenshots (as I don't have [...]

 
[image]

Clicking the top button redirects to this page:

https://entrez.enphaseenergy.com/authorize?code_challenge=4bbwml7hgn1MJQWf9nLtyzx892MxjA1RPsPaNimflPw&client_id=envoy-ui-client&redirect_uri=https://192.168.1.69/auth/callback&scope=########&response_type=code&code_challenge_method=S256

and then this page after auth:

https://192.168.0.5/auth/callback?code=776e446b-c93f-4946-9efd-1ed557a6199f

Let me know if I can provide more details! Thanks

enkrypt3d, 2022-01-08:
[image]

gtdiehl, 2022-01-09:
I'm not too sure about the code after I retrieve the token. The below code scrapes a token from the Enphase site, just have to replace the ```username```, ```password```, and Envoy IP Address.

After getting the token I'm not sure 
[image]

```
import requests
from bs4 import BeautifulSoup

payload = {"username": "aaaaa", "password": "bbbbb"}

session = requests.session()
r = requests.post("https://entrez.enphaseenergy.com/login", data=payload)

payload = {"uncommissioned": "true", "Site": ""}
cookies = requests.utils.cookiejar_from_dict(
    requests.utils.dict_from_cookiejar(r.cookies)
)

r = requests.post(
    "https://entrez.enphaseenergy.com/entrez_tokens", data=payload, cookies=cookies
)
soup = BeautifulSoup(r.text, features="html.parser")
jwt = soup.find("textarea").contents[0]
print("Token: " + jwt)

payload = {"Authorization": "Bearer " + jwt}
r = requests.get("https://x.x.x.x/auth/check_jwt")

r = requests.get("https://x.x.x.x/production.json")
print(r.text)
```

gtdiehl, 2022-01-09:
@enkrypt3d Can you run the code against your Envoy?

It'll probably give an error when retrieving the production page.

enkrypt3d, 2022-01-09:
How do I do that? I'm not a python expert so I'd need some guidance.

After authenticating, I'm able to hit that production.json:

{"production":[{"type":"inverters","activeCount":34,"readingTime":0,"wNow":0,"whLifetime":149221},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1641697963,"wNow":-0.0,"whLifetime":2547550.432,"varhLeadLifetime":9.708,"varhLagLifetime":1454333.844,"vahLifetime":3614484.427,"rmsCurrent":6.029,"rmsVoltage":248.535,"reactPwr":742.428,"apprntPwr":749.07,"pwrFactor":0.0,"whToday":49074.432,"whLastSevenDays":2498476.432,"vahToday":59799.427,"varhLeadToday":1.708,"varhLagToday":16365.844}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1641697963,"wNow":-57.773,"whLifetime":3317154.764,"varhLeadLifetime":2184853.326,"varhLagLifetime":1462122.356,"vahLifetime":4204961.002,"rmsCurrent":-7.788,"rmsVoltage":248.509,"reactPwr":-1905.093,"apprntPwr":-1935.294,"pwrFactor":-1.0,"whToday":40856.764,"whLastSevenDays":3276307.764,"vahToday":41020.002,"varhLeadToday":23112.326,"varhLagToday":16492.356},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1641697963,"wNow":-57.773,"whLifetime":1525239.582,"varhLeadLifetime":2184843.618,"varhLagLifetime":7788.512,"vahLifetime":4204961.002,"rmsCurrent":13.817,"rmsVoltage":248.482,"reactPwr":-1162.665,"apprntPwr":1716.703,"pwrFactor":-0.03,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

enkrypt3d, 2022-01-09:
Also I think we can ignore the token authentication method. the top option
only uses the enlighten manager login....

gtdiehl, 2022-01-09:
Copy the above Python code into a text editor like Notepad and change the username, password, and IP Address placeholders. Save the file as `tokentest.py`. From the command line, change directory to where you saved the Python file and run `python tokentest.py`

Assuming you have Python installed

enkrypt3d, 2022-01-09:
***@***.***:~# python tokentest.py
Traceback (most recent call last):
  File "/root/tokentest.py", line 19, in <module>
    jwt = soup.find("textarea").contents[0]
AttributeError: 'NoneType' object has no attribute 'contents'

gtdiehl, 2022-01-09:
I wasn't expecting it to fail there. I added some print statements to the output. Not sure if you want to post the output though. You can email it to me if you feel more comfortable.

enkrypt3d, 2022-01-09:
Got it - sent you the output in your inbox

gtdiehl, 2022-01-12:
I'm working on the changes but have run into a wall. When using the ```httpx.AsyncClient()``` it does not seem to work the same as when using ```httpx.Client()```. When trying to get the token from Enphase I can login, but when I goto the ```TOKEN_URL``` page I get an Authentication error. Maybe some session info is not being used/saved?

My current code is located at:
https://github.com/gtdiehl/envoy_reader/blob/token_firmware/envoy_reader/envoy_reader.py

enkrypt3d, 2022-01-12:
So I only need to replace the envoy_reader.py file? Or are there other
changes needed?

gtdiehl, 2022-01-12:
No don't replace it. It can be run by itself from the command line. But at this point nothing to test as it will fail. I have to investigate how to use the async client and have it behave the same as the sync client that was in the proof of concept code.

enkrypt3d, 2022-01-14:
# python envoy_reader.py
Enter the Envoy IP address or host name, or press enter to use 'envoy' as default: 192.168.196.1.69
Enter the Username for Inverter data authentication, or press enter to use 'envoy' as default: #######@gmail.com
Enter the Password for Inverter data authentication, or press enter to use the default password: ########
Reading...
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:__main__:HTTP GET Attempt #1: http://192.168.196.1.69/production.json
DEBUG:__main__:HTTP GET Attempt #2: http://192.168.196.1.69/production.json
DEBUG:__main__:HTTP GET Attempt #3: http://192.168.196.1.69/production.json
DEBUG:__main__:HTTP GET Attempt #1: http://192.168.196.1.69/api/v1/production
DEBUG:__main__:HTTP GET Attempt #2: http://192.168.196.1.69/api/v1/production
DEBUG:__main__:HTTP GET Attempt #3: http://192.168.196.1.69/api/v1/production
DEBUG:__main__:HTTP GET Attempt #1: http://192.168.196.1.69/production
DEBUG:__main__:HTTP GET Attempt #2: http://192.168.196.1.69/production
DEBUG:__main__:HTTP GET Attempt #3: http://192.168.196.1.69/production
Traceback (most recent call last):
  File "/root/envoy_reader.py", line 716, in <module>
    TESTREADER.run_in_console()
  File "/root/envoy_reader.py", line 588, in run_in_console
    data_results = loop.run_until_complete(
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/root/envoy_reader.py", line 230, in getData
    await self.detect_model()
  File "/root/envoy_reader.py", line 318, in detect_model
    raise RuntimeError(
RuntimeError: Could not connect or determine Envoy model. Check that the device is up at 'http://192.168.196.1.69'.

It's not looking at HTTPS?

gtdiehl, 2022-01-14:
For the new firmware, additional command line arguments have to be used. In my previous comments above two examples are shown for the Uncommisioned and Commissioned Envoy token.

enkrypt3d, 2022-01-14:
Sent you an email

gtdiehl, 2022-01-19:
I've released the `envoy_reader` library with support for firmware 7.x.x.

Also I have made changes to the Home Assistant `enphase_envoy` sensor integration. I'll leave this Issue open until the HA side has been completed.

Before submitting the PR on the HA side it should probably be tested out. The sensor code is located at:
https://github.com/gtdiehl/core/tree/envoy_new_fw/homeassistant/components/enphase_envoy

Copy over the above directory and replace it in your HA server instance.

I don't have an Envoy running this firmware so I would say the more testing the better.
Thanks!

enkrypt3d, 2022-01-19:
@gtdiehl thank you! So glad this is working again! Hopefully the tokens get automatically refreshed. I'll let you know how it looks tomorrow.

gtdiehl, 2022-01-19:
It looks like the code when used in Home Assistant, is hitting the Enphase server every polling cycle. Before wide usage this should probably be resolved! I'll give an update once this is resolved.

IshmaelObeso, 2022-01-19:
@gtdiehl I'm running this on my HA instance now, looks like it's working. Hit me up if you want to test anything or check on specific issues :)

gtdiehl, 2022-01-22:
@IshmaelObeso Did you setup the Envoy using Commissioned or Uncommisioned?

Also I just updated the HA integration code yesterday so that it should only hit the Enphase server when the token expires. In the earlier code it was hit them every polling cycle!

IshmaelObeso, 2022-01-22:
Commissioned, it works fine so far only thing is it seems to forget the
configuration every ~12 hours and I have to reinput the credentials.
I'll update the code after the weekend and test it out and let you know if
anything pops up. My system also has batteries if you ever want to do
testing on those.

 Ishmael Obeso | UCLA Neuroscience 2019 | Tel: (626)-703-2047

enkrypt3d, 2022-01-27:
> Commissioned, it works fine so far only thing is it seems to forget the configuration every ~12 hours and I have to reinput the credentials. I'll update the code after the weekend and test it out [...]

Not having that issue here. Mine also has batteries.

gtdiehl, 2022-01-27:
Ok great thank you! I’ll look into the credential issue. It seems others
are seeing the same issue so I hope I can track it down

IshmaelObeso, 2022-01-27:
@gtdiehl Sounds good, hit me up if you need me to pull logs or anything. The configuration issue seems to be happening more often after I updated the code to your newest version.

djk121, 2022-02-09:
Commissioned, with batteries. I pulled the enphase_envoy directory @gtdiehl mentioned above on Saturday, that caused the integration to work for me. It worked until last night, when it hit an auth error:

2022-02-08 22:53:25 ERROR (MainThread) [homeassistant.components.enphase_envoy] Authentication failed while fetching envoy Envoy data: Client error '401 Unauthorized' for url 'https://192.168.2.85/api/v1/production/inverters'

djk121, 2022-02-17:
I've been watching how @gtdiehl's patch above works for about 10 days now, and I think there are two things that cause things to fail:
1. If there are any network changes on the envoy, a new JWT is required.
2. Because python, sometimes there's a big enough delay between when token validity is checked and when the request is sent to the envoy.

Both of these result in the envoy sending back a 401, which turns into an exception to HA.

https://github.com/jesserizzo/envoy_reader/pull/90 to retry these (after fetching a new JWT) instead.

enkrypt3d, 2022-02-18:
> I've been running with #90
> <https://github.com/jesserizzo/envoy_reader/pull/90> for the last couple
> of days, and it does seem to keep the integration working for me. I'm not
> sure *why* the Envoy is throwing the 401s it is occasionally handing out,
> but reacting to them by fetching a new token and retrying seems to paper
> over the issue.

I made the mistake of upgrading to 2022.01 and broke everything on the
enphase integration. I have downgraded the HA Core back to 2021.12.10 which
was working, and the data is coming in but it never gets to the point of
creating the entities and finally bombs out saying integration can't be
setup.

Do I need to downgrade supervisor as well? I'm able to see the data coming
into the HA logs but eventually the integration times out.

Logger: homeassistant.setup
Source: setup.py:291
First occurred: 5:01:04 PM (1 occurrences)
Last logged: 5:01:04 PM

Unable to prepare setup for platform enphase_envoy.sensor: Platform not
found (No module named 'homeassistant.components.enphase_envoy.sensor').

[image: image.png]

enkrypt3d, 2022-02-20:
I believe I have fixed most of my issues with this integration, but getting this error when I Try to add consumption? The production numbers are working well though: 

[image]

briancmpbll, 2022-02-24:
> Before submitting the PR on the HA side it should probably be tested out. The sensor code is located at: https://github.com/gtdiehl/core/tree/envoy_new_fw/homeassistant/components/enphase_envoy
> 
> Copy over the above directory and replace it in your HA server instance.
> 

Pardon my ignorance, but where does this directory get copied into? I'm running hass os on a raspberry pi and I don't see a directory that would match this one in the file system.

I just got a enphase system installed with 7.x.x and would like to help test this.

enkrypt3d, 2022-02-24:
```/usr/src/homeassistant/homeassistant/components/enphase_envoy/```

This is the path in my docker container. yours may be a little different. be sure to empty that folder out and ensure u have all files:
```
wget https://raw.githubusercontent.com/gtdiehl/core/envoy_new_fw/homeassistant/components/enphase_envoy/__init__.py
wget https://raw.githubusercontent.com/gtdiehl/core/envoy_new_fw/homeassistant/components/enphase_envoy/config_flow.py
wget https://raw.githubusercontent.com/gtdiehl/core/envoy_new_fw/homeassistant/components/enphase_envoy/const.py
wget https://raw.githubusercontent.com/gtdiehl/core/envoy_new_fw/homeassistant/components/enphase_envoy/manifest.json
wget https://raw.githubusercontent.com/gtdiehl/core/envoy_new_fw/homeassistant/components/enphase_envoy/strings.json
wget https://raw.githubusercontent.com/gtdiehl/core/envoy_new_fw/homeassistant/components/enphase_envoy/sensor.py

wget https://raw.githubusercontent.com/gtdiehl/core/envoy_new_fw/homeassistant/components/enphase_envoy/translations/en.json
```

enkrypt3d, 2022-02-24:
Anyone know how to monitor batteries with this integration?

del13r, 2022-02-26:
@enkrypt3d 
Have you tried checking this link http://envoy/production.json 
There is a storage section of that json file hosted on the Envoy.
Mine is blank as I don’t have a battery.

"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

del13r, 2022-02-26:
@enkrypt3d 
also try http://envoy.local/ivp/ensemble/inventory
where you will see information like this
“percentFull”: 3, “temperature”: 13, “maxCellTemp”: 14,

enkrypt3d, 2022-02-26:
Yes I'm able to see these values, just trying to figure out how to create
entities in home assistant with these batteries. I have 3 encharge 10's (9
total batteries)

michaelgaultjr, 2022-03-10:
> `/usr/src/homeassistant/homeassistant/components/enphase_envoy/`
>
> This is the path in my docker container. yours may be a little different. be sure to empty that folder out and ensure u have all [...]

Hello, I've been using this integration for a few days on two combiners, one running D5.x.x firmware, the other running a D7.x.x firmware in "Commissioned" mode, and it's mostly been working fine except for every morning around 4am-5am the D7.x.x combiner integration disconnects with a `401 unauthorized` error.

rjnca, 2022-03-24:
Has there been any movement on this?  I'm still getting the "unexpected error" message both with the core integration and with the custom component.  Is there something I'm missing?

enkrypt3d, 2022-03-24:
Did you try deleting the integration, wget'ing the files, then restart HA,
then re-add the integration? You have to login with your enlighten account
now....

michaelgaultjr, 2022-03-24:
> Has there been any movement on this? I'm still getting the "unexpected error" message both with the core integration and with the custom component. Is there something I'm missing?

I've been using [the branch by @DanBeard](https://github.com/jesserizzo/envoy_reader/pull/91) and it's been working perfectly for me, the only issues I've experienced are issues with Enphase combiners themselves. Just keep in mind you may have to delete version in the python cache as described here https://github.com/jesserizzo/envoy_reader/pull/91#issuecomment-1065942291

newburns, 2022-04-01:
> `/usr/src/homeassistant/homeassistant/components/enphase_envoy/`
>
> This is the path in my docker container. yours may be a little different. be sure to empty that folder out and ensure u have all [...]

Is it possible to do this with HomeAssistant OS? There is no components or homeassistant folder that I can find. I'm guessing it's structure is different in the OS version (HAOS).

I am trying to do this through custom_components folder. Not sure if I need to add something to configuration.yaml for that

enkrypt3d, 2022-04-01:
You should be able to do a "locate enphase_envoy" and it should find the location. I'm using docker on a raspberry Pi 4. It's the default integration, not a custom component.

@gtdiehl any way to get the batteries added like it is in this repo? 

https://github.com/DanBeard/enphase_envoy

I tried his but it didn't work for me for some reason.

briancmpbll, 2022-04-03:
Hass OS doesn’t work the same as a docket install. As far as I can tell, the home assistant source directories are all locked away somewhere that is not user accessible. I haven’t been able to find a way to use this on my hass os install.

briancmpbll, 2022-04-03:
I packaged this up as a HACS custom component for use with HASS OS: https://github.com/briancmpbll/home_assistant_custom_envoy

@newburns take a look at see if this will work for you.

peterwalkowiak, 2022-04-12:
This looks like it has been officially adopted now. HACS component stopped working and the official integration now works! Thank you all!

enkrypt3d, 2022-04-16:
Peterwalkowiak, Is there some documentation that shows this integration is
now merged with HA Core? Can't seem to find anything.

peterwalkowiak, 2022-04-16:
> Peterwalkowiak, Is there some documentation that shows this integration is now merged with HA Core? Can't seem to find anything.
> […](#)

Something happened with my HA install and the custom showed up as the core for me. I deleted my comment because I realized I was wrong shortly after posting. Still crossing my fingers!

anthony4000, 2022-04-19:
I have two Envoys. 1 running version 5.05 and another with version 7.3. This version (HACs) works great with v7.3. However, now I'm unable to use it with v5.05. Integration just says retrying setup. I'm using IP, (user)envoy, serial. If I remove HACs component and revert to core, v5.05 works with no issues.

Log:

> Logger: custom_components.enphase_envoy
Source: custom_components/enphase_envoy/envoy_reader.py:721
Integration: Enphase Envoy
First occurred: 7:24:04 PM (11 occurrences)
Last logged: 7:31:48 PM
Unexpected error fetching envoy Envoy 202140124443 data: 0
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 190, in _async_refresh
    self.data = await self._async_update_data()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 150, in _async_update_data
    return await self.update_method()
  File "/config/custom_components/enphase_envoy/__init__.py", line 63, in async_update_data
    battery_data = await envoy_reader.battery_storage()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 721, in battery_storage
    if len(ensemble_json) > 0 and "devices" in ensemble_json[0].keys():
KeyError: 0

djk121, 2022-04-29:
Is this going to get fixed? There are several PRs that fix the authentication errors reported in this issue, but they've been sitting there for ~months with no comment/response.

enkrypt3d, 2022-04-29:
https://github.com/briancmpbll/home_assistant_custom_envoy I ended up using
this and it has been working well for me.

djk121, 2022-04-29:
:(

So is this abandoned? Is it worth sending a PR to the HA folks to switch away from this to that?

jesserizzo, 2022-04-30:
Hi all, sorry I've been pretty absent from this. Can anyone on firmwares other than 7.0.43 and 5.0.49 test this out to make sure it doesn't break anything?

sergeknystautas, 2022-05-02:
Hi! I spent the weekend walking through a different token issuing approach for 7.X firmware.  I just had my system installed a week ago and was pretty miffed that nothing was working with Home Assistant.

I took a very different approach than this PR takes.  First off, my enphase admin screen does not have a radio buttons for commissioned or uncommissioned, so I'm not sure how this would work.  My approach ports the JS implementation provided by Enphase when you hit the https://envoy.local/ URL.  Their implementation for getting local JWT tokens works like this...

1. Locally generate 40-character random string, and SHA256 hash it.
2. Send user to https://entrez.enphaseenergy.com/login with a callback parameter and the SHA256 hash.
3. If the user enters successful credentials, it creates a tracking code and sends you back to https://envoy.local/auth/callback
4. Note to get this auth flow working, the Python implementation does not have to hit https://envoy.local/auth/callback.  This is just an HTML page that generates an XHR call for the next step, that Python does need to call.
5. Call https://envoy.local/auth/get_jwt with your original random string and the tracking code.  This returns a JSON of an access token.

I got this flow all working, with some python JWT parsing to check if the token has expired and automatically refresh the token if needed.  Note you have to ignore the SSL verification and the warnings from that because the https://envoy.local/ is a self-signed cert.  It's fundamentally just 2 HTTPS calls though... 1 to login, and 1 to get_jwt.

I think the benefit of my approach is that you just need to configure your username/password, and this will handle continually refreshing the JWT as needed.  Also it involves no cookies or HTML parsing, which I think will make it less likely to break over time.  And ultimately then you can use that JWT then for whatever API calls you want.

Note I'm doing this against D7.3.75 (47f7da)

[24 more replies]